wuxx/nanoESP32-C6 #3: Any example pinout reference?

ckcr4lyf, 2023-11-22:
I just received my board from Taobao. I was wondering if there is some example refence , so I can use it with PlatformIO or similar.

Example: https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/_images/esp32-c6-devkitc-1-pin-layout.png

Or is the number on board directly correspond the pin number for the module, e.g. (https://www.espressif.com/sites/default/files/documentation/esp32-c6-wroom-1_wroom-1u_datasheet_en.pdf)
[image]

ckcr4lyf, 2023-11-22:
Also wondering if it is compatible with the Arduino framework, e.g. https://github.com/espressif/arduino-esp32

wuxx, 2023-11-27:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/index.html

wuxx, 2023-11-27:
you can check the schematic, it's GPIO8
https://github.com/wuxx/nanoESP32-C6/blob/master/hardware/nanoESP32C6.pdf