wuxx/nanoESP32-C6 #7: Can´t connect to board to flash

henmartins27, 2024-07-19:
Hello,
I followed the demo guide to verify my board on both Windows 11 and Linux (Ubuntu). In both cases I get the following problem:
```
Serial port /dev/ttyACM0
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-C6: No serial data received.
```
On Windows I use serial port COM3.

How to fix this behavior? Thanks

wuxx, 2024-07-19:
Hello, what is the complete command you entered

henmartins27, 2024-07-19:
@wuxx I used the same command as the demo:

`$git clone https://github.com/espressif/esp-idf.git
$cd esp-idf && ./install.sh && . ./export.sh
$cd examples/get-started/blink
$idf.py set-target esp32c6
$idf.py -p /dev/ttyACM0 flash monitor`

I now tried running it on Arduino IDE (with the recommended config) and I faced the exact same error.

wuxx, 2024-07-19:
Please try manually entering download mode, holding down the BOOT button to power on, and then executing the download command

henmartins27, 2024-07-19:
@wuxx Tried this (hold down BOOT, then press RST then release both) and still has same issue. Is it possible that my board is defective?

henmartins27, 2024-07-19:
Now I also tried on macOS (completely different computer) and found the same error...

wuxx, 2024-07-19:
The default program is a flashing light. Can you observe it? In addition, there are two USB ports on the board. Which one did you plug in?

henmartins27, 2024-07-19:
No, the light never turns on. I’m using the port on the left - CH343. The other one is never recognized by the computer. 

Just to note, the error is different when the board is not plugged in, so the computer is seeing something, it’s just not managing to flash it.

wuxx, 2024-07-22:
Can you check the module to see if there is any solder joint problem?

Bart56, 2024-09-26:
I'm plugging into the CH343 port using a win 7 laptop and it won't install a driver. I've held the boot button down until after plugged in but same error. This is a nanoESP32-C6 V1.0 board. What do I need to do to get it to recognize it?