wuxx/nanoESP32-C6 #9: Onboard LED not working

lukechadwick, 2024-10-30:
The board is working properly via arduino and I can address the pins properly.

I purchased several of these boards and they seem to have the same issue.  Using a multimeter I can see voltage going to the LED as it turns on/off but it doesn't light up.  On one of the boards it did light up when I pushed down on it a little and cycled some colors before turning off again and hasn't worked since.

Any ideas?

wuxx, 2024-10-31:
Hello, where did you purchase it from? Each one has been tested before delivery, and the default program is flashing lights

lukechadwick, 2024-10-31:
Hi I got it from AliExpress and the description has the link to this GitHub
https://a.aliexpress.com/_msEM9cG

wuxx, 2024-10-31:
This is not our store. What you purchased is a pirated copy of ours.
this is our shop, thanks.
https://miusecntech-muselab.aliexpress.com/store/1101496384

lukechadwick, 2024-10-31:
Okay I will report them on AliExpress and try your store in the future.  Thank you.

lukechadwick, 2024-11-03:
Hi @wuxx are you still selling these? I can't find them in the store and the link on github is 404

wuxx, 2024-11-03:
Hello, can you visit this link?
https://miusecntech-muselab.aliexpress.com/store/1101496384?spm=a2g0o.detail.0.0.58980zS90zS9gu

closed 2024-11-08 by lukechadwick