wuxx/nanoESP32-C6 #11: WS2812B should be powered from 5V, not 3V3

tjfsteele, 2025-03-18:
The official devkit boards use 5V for the WS2812B, and the datasheet for the WS2812B shows a minimum supply voltage of 3V6. Also, the input data signal should not go higher than the supply voltage, which it will on your design as you're converting it to 5V logic level.

I have checked one of your V1.0 boards and confirmed the WS2812B is indeed being supplied with 3V3 as per your schematic.

wuxx, 2025-03-19:
Hello, yes, you are right, the next version will optimize this problem

closed 2025-04-19 by wuxx

tuxcrafter, 2026-05-03:
@Gunni thank you again for sharing your code, I was using a different board variant but it all did not make a difference, no led light at all.