wuxx/nanoESP32-C6 #12: Full & Complete pinout diagram for this board please

Anandababu, 2025-03-30:
Its very difficult to find the pinout references without this board specific diagram. All the references are pointing to generic boards and chips. 

Is it possible to publish a full and complete pinout diagram for this board please?

tvixen, 2025-04-19:
I have tried to track down any documentation on this board, but without success, 
It's the worst documented board ever. They did even care to copy any of the original onto their /DOC folder.

wuxx, 2025-04-19:
Hello, actually just having a schematic is enough for development, because the ESP chip has IOMUX function, and IO and hardware functions can be mapped

wuxx, 2025-04-19:
This is the official SDK of espressif
https://github.com/espressif/esp-idf

tvixen, 2025-04-19:
Yes, it is appreciated, but the documentation on this board is really poor. Also, the hardware does not follow the standard on other boards. 
But we get what we pay for.... Next time we just have to pay a little more.

MrBran4, 2025-04-27:
Next time I'll be paying for one that's not a knockoff - they're not even that much more expensive!

Anyway, if it helps @tvixen the pinout seems to more-or-less match the ESP32-C6-WROOM-1 with two extra pins at the bottom near the usb's and skipping a pin on the right

Found this in the schematic dir of this repo, which matches the markings of the board itself

[image] 

The above seems to match the pinout from the official espressif docs (below) but with the differences:
- Extra 5v lines (guessing just vbus) at the bottom next to the USBs
- The NC pin on the chip is skipped on the board, replaced with an extra gnd at the bottom

[image]

From there, things like spi/etc can be figured out from the espressif docs. Or at least that's my plan - I'll report back here if I manage to burn my house down trying

tvixen, 2025-04-27:
Thanks for the information.... most appreciated, and yes next time I will be more aware of the quality.

Low-Code-Platform, 2025-07-22:
Any idea how to call the onboard LED? I’m using the adafruit neopix library in Arduino IDE on pin 8, but no luck.

Gunni, 2025-08-07:
Just wanted to mention that I was able to control the LED from ESPHome using the following:

```yaml
light:
  - platform: esp32_rmt_led_strip
    internal: True
    rgb_order: GRB
    pin: GPIO8
    num_leds: 1
    chipset: ws2812
    id: esp32_rgb_led
    restore_mode: ALWAYS_OFF
    default_transition_length: 0ms
    initial_state: 
      color_brightness: 11%
      brightness: 11%

# Blink demo?
interval:
  - interval: 
      seconds: 1
    then: 
      - light.toggle: esp32_rgb_led
```

`working > correct` but if something is set wrong, let me know.

p.s. this LED at 100% brightness is really blindingly bright, I typically run them at 11% brightness but I have not had `initial_state` work, i need to set it `on_boot`.

gasiepgodoy, 2025-08-12:
> Just wanted to mention that I was able to control the LED from ESPHome using the following:
> 
> light:
>   - platform: esp32_rmt_led_strip
>     internal: True
>     rgb_order: GRB
>     pin: GPIO8
>     num_leds: 1
>     chipset: ws2812
>     id: esp32_rgb_led
>     restore_mode: ALWAYS_OFF
>     default_transition_length: 0ms
>     initial_state: 
>       color_brightness: 11%
>       brightness: 11%
> 
> # Blink demo?
> interval:
>   - interval: 
>       seconds: 1
>     then: 
>       - light.toggle: esp32_rgb_led
> `working > correct` but if something is set wrong, let me know.
> 
> p.s. this LED at 100% brightness is really blindingly bright, I typically run them at 11% brightness but I have not had `initial_state` work, i need to set it `on_boot`.
> 
> p.p.s. While it's awesome to get an RGB led, it's annoying that such LEDs are NOT supported at all as [Status LED](https://esphome.io/components/status_led.html) in ESPHome. [esphome/feature-requests#2230](https://github.com/esphome/feature-requests/issues/2230)

Which version is your board? It seems that v1.0 has a circuit problem with the RGB led (https://github.com/wuxx/nanoESP32-C6/issues/11). Tks

sbf-, 2026-04-17:
[image]

Found it [here](https://www.reddit.com/r/esp32/comments/1lfpi11/just_incase_anybody_needs_it_esp32c6_pinout/)