wuxx/nanoESP32-C6 #15: How do you call the onboard led?

Low-Code-Platform, 2025-07-22:
How do you call the onboard LED? What’s the PIN number? I’m calling it on pin 8, using the adafruit neopix library, but no luck.

wuxx, 2025-07-23:
hello, it's WS2812 on Pin 8, you can try this demo
https://github.com/wuxx/nanoESP32-C6/tree/master/demo

wuxx, 2025-10-23:
try this demo https://github.com/wuxx/nanoESP32-C6/tree/master/demo/blink

bayuTam, 2025-11-22:
I think there is a mistake in the wiring (https://github.com/wuxx/nanoESP32-C6/blob/master/hardware/nanoESP32C6.pdf). In the WS2812B datasheet  (https://cdn-shop.adafruit.com/datasheets/WS2812B.pdf) the working voltage is +3.5 to +5.3 volts, but in the schematic the VDD pin is connected to 3.3V, this voltage does not meet the requirements, and after I checked with a multimeter the DIN pin is connected directly to the 5V pin, it seems there is no 10K R-pullup